pace-neutrons/Horace #7: Tobyfit with improved instrument classes

tgperring, 2019-04-29:
New instrument classes follow  post R2008 Matlab class definitions and are more flexible than old instrument definitions

abuts, 2021-01-27:
Tobyfit with new classes is there and operational.
New instrument classes tests: `Herbert\_test\test_instrument_classes`
Fitting with new instruments: `Horace\_test\test_tobyfit`. Tobyfit can not work without instrument and all these tests are testing new instruments. 
Saving loading new instruments with sqw file: `Horace\_test\test_sqw_file\test_faccess_sqw_v3.m` 
Setting instrument on sqw object.: `Horace\_test\test_sqw_file\test_instrument_methods.m` `Horace\_test\test_sqw_file\test_set_instrument_data_2.m` 

The only issue may be related to this is removing legacy Tobyfit, which may use some old data, but it certainly have its own ticket

cmarooney-stfc, 2021-01-27:
Tested this against test_tobyfit and test_instrument_class scripts and confirmed it is using the new classes.

closed 2021-02-10 by cmarooney-stfc