pace-neutrons/Horace #22: Optimize gen_sqw

nickbattam-tessella, 2019-11-11:
This tool must be optimized for multi-thread and multi-core execution.

Identified by Alex as benefiting from parallelization

nickbattam-tessella, 2019-11-15:
Depends on #55

nickbattam-tessella, 2020-04-17:
Depends on #69

nickbattam-tessella, 2020-05-20:
Notes from @abuts  

The optimized parallel implementation is there and works for users.
gen_sqw contains 3 parallel algorithms. 

1) calculate tmp files 
-- massively parallel task without interprocess communications. (only log progress) 50% of time
--  implemented in- @gen_sqw_files_job  works with any parallel framework except I did not tried CPP_communicator as its incomplete. Should work with it too. 

2) accumulate headers 
-- reasonably parallel communications with parallel reader and partial image exchange over mpi. (5min max. ) Short job, so normally no point to launch separate cluster for it, so its parallelized because cluster is running from calculate tmp.  Classical texbook MPI job. 
 -- implemented in @accumulate_headers_job  
-- works with filebased and Matlab MPI 
-- CPP_communicator needs blocking messages implemented to process it (BLOCKED)
-- problem with filebased messages have been recently discovered on iDaaaS, so reliably works only with Matlab MPI. 

3) combine pixels -- parallel multiple reader-single writer application with extensive interprocess communications.  50% of time. 
-- implemented in @combine_sqw_pix_job  and its a prototype. It runs on @abuts computer with file-based and Matlab MPI, but fails on Jenkins windows (Linux works).  New file-based messages started to fail this job on windows machine, so fixing it become much easier.  
-- until we are running our parallel applications on a single node I have sort_pixels_by_bins.mexw64  which is parallel multiple reader single writer application using threads. Its very efficient, so its currently used on iDaaaS and was gobbling all bandwidth, any filesystem I tried to, can provide.

hsautessella, 2020-05-26:
This document contains information on how parallel gen_sqw works https://github.com/pace-neutrons/Herbert/blob/master/documentation/ParallelFrameworkDesign.docx

nickbattam-tessella, 2020-11-04:
Blocked on: https://github.com/pace-neutrons/Horace/issues/68