pace-neutrons/Horace #25: Optimize TobyFit algorithm

nickbattam-tessella, 2019-11-11:
This tool must be optimized for multi-thread and multi-core execution.

Identified by Alex as benefiting from parallelization.

This is not embarrassingly parallelizable as there is cross talk between sections of data.

nickbattam-tessella, 2019-11-15:
Depends on #55

nickbattam-tessella, 2019-11-19:
From Toby:

The current Multifit has the functionality that we need, but (1) was never written with multicore (and certainly not distributed) computing in mind, (2) plug-and-play for the fitting method (i.e. use genetic algorithms, global minimisation, whatever...). Multifit also has a nested fit functions capability that has made it easy, for example, to wrap functions that calculate models for S(Q,w) with resolution convolution - in fact that uses two levels of nesting because the models for S(Q,w) is wrapped by the function sqw_eval.
I think it is probably worth looking at what has been implemented in Mantid for fitting - I had many long discussion with Nick Draper back in the day when fitting was being incorporated into Mantid, particularly about the idea of local and global fitting models. Regarding multicore and multinode fitting, I would have thought that there is work in the public domain to be explored.

oerc0122, 2023-06-20:
Done with parallelise-tobyfit merge

closed 2023-06-20 by oerc0122