pace-neutrons/Horace #27: Technology Test: Python wrapping Matlab wrapping multi-threaded C++

nickbattam-tessella, 2019-11-11:
Initial design discussions suggest the Python API may be created by creating a Python API to a compiled (Matlab) Horace implementation.

This ticket is to verify this process would work and function as expected in a multithreaded environment.

mducle, 2019-11-11:
I had a go at writing a Python wrapper for Horace https://github.com/mducle/hugo but it doesn't really work because the main `sqw` class is not a handle class and the matlab-python bridge automatically converts it to a `dict` and cannot convert back to the corresponding matlab class.

nickbattam-tessella, 2019-11-11:
Thanks @mducle.

I'm hoping the changes we're planning for a rewrite of the Matlab SQW object (#21) will simplify this mapping.

abuts, 2019-11-11:
@mducle Are you talking about handle classes or modern day Matlab reference classes?
We have not desided on the switch to the handle classes, as the user experience will be substantially different. 

If the only class type acceptable for compilation into Python is the handle class, we need substantial Horace overhaul. It looks like we can do it, but I the desicion has to be taken now.

mducle, 2019-11-11:
@abuts I'm not sure - I didn't test it much. I think reference classes would be fine but I was going with the fact that the SpinW python interface works and it uses handle class. To be honest, I don't think there is going to be *that* much difference in the user experience - I don't really think a lot users regularly use the copy assignment behaviour and we certainly don't really advertise it much in the tutorials.

abuts, 2019-11-11:
@mducle @nickbattam-tessella  @hsautessella @tgperring   If this is the case, its the critical job which has to be done first. And it a good start-up job to replace ~50+ classes from reference to handle and modify all unit tests to accept classes differently
And to add deep-copy methods. 

Not sure the managment and advanced users  have approved this, so I would ask Toby's opinion first.

mducle, 2022-03-21:
As mentioned in the linked issue 48, handle classes are not required - in principle a new `classdef` type class is needed for the Matlab-Python bridge but the `pace-python` implementation has a [work-around](https://github.com/pace-neutrons/pace-python/blob/main/src/thinwrapper.m).

Multithreaded `mex` files work fine in the compiled Matlab code.

closed 2022-03-21 by mducle