pace-neutrons/Horace #31: Dependency management

nickbattam-tessella, 2019-11-11:
C++ build will include dependencies on a number of third-party libraries, e.g. boost, mpi, hdf5, testing framework...

A plan for managing this should be created and filed as an ADR and implemented.

nickbattam-tessella, 2019-11-27:
MATLAB enforces strict versioning of libraries, often not-current releases.

As part of this investigation into dependency management we must find a solution to map the correct library to the required MATLAB version

e.g.
- HDF5 1.8.12 for <current releases of MATLAB>
- HDF5 1.8.6 for <legacy releases of MATLAB> TODO: confirm WHICH version
- build against HDF5 libraries found on the MATLAB path for Linux
- build against HDF5 libraries managed locally for Windows as the MATLAB doesn't include the static libraries
- FAIL the build if the required headers or libraries matching the found MATLAB version don't exist

There is no simple reference on the Mathworks site for which MATLAB releases are shipped with which library.

hsautessella, 2019-11-28:
As part of this ticket, we should also look into how we deal with other PACE dependencies and modelling codes, e.g. [Herbert](https://github.com/pace-neutrons/Herbert) and [Euphonic](https://github.com/pace-neutrons/Euphonic)

nickbattam-tessella, 2020-01-23:
Time boxed based document decisions based on current understanding

hsautessella, 2020-02-25:
Closed by #132

closed 2020-02-25 by hsautessella