pace-neutrons/Horace #32: C++ unit testing

nickbattam-tessella, 2019-11-11:
Identify the most appropriate C++ testing library and migrate any existing tests to this.

Outputs:
- ADR documenting selection
- Implementation of at least one test

hsautessella, 2019-11-15:
Googletest was identified as the preferred choice for testing framework.

1) It is in active development - receiving bug fixes
2) It is in use by other teams within ISIS (IBEX, Mantid uses it for the built-in mocking framework)
3) It is required by [Google Benchmark](https://github.com/google/benchmark#requirements) which may become useful in the future.

The fact that it was in use by other teams within ISIS outweighed the fact that Catch2 is more light-weight.

abuts, 2019-11-15:
Bearing in mind that the only unit tests currently existing for C code are the tests for CombineSQW, do not see any problem in changing to this. 

Matlab code tests should be still written on the basis of matlab_xunit  tests framework

closed 2020-01-02 by nickbattam-tessella