pace-neutrons/Horace #35: Manage version number for builds and releases

nickbattam-tessella, 2019-11-11:
Create managed release number to replace legacy SVN based numbering:
m: major
n: minor
o: bugfix
p: datetime (yyyymmdd-hhmmss)

p to be added at automatically at buildtime; m,n,o from developer managed version text file.

nickbattam-tessella, 2020-01-20:
`horace_init` currently prints two SVN-style revision numbers and datetime:
```
! Matlab  code: $Revision:: 1758 ($Date:: 2019-12-16 18:18:50 +0$) !
! Mex files   : $Revision:: 1757 ($Date:: 2019-12-05 14:56:06   $) !
```

The majority of Horace and Herbert source files include a revision number at the end of the header:
```
% $Revision:: 1758 ($Date:: 2019-12-16 18:18:50 +0000 (Mon, 16 Dec 2019) $)
```
injected by `horace_core\admin\horace_version.m` and `herbert_core\admin\herbert_version.m`.

Since the move to Git the version should now be datetime, sha or tag-based.

nickbattam-tessella, 2020-01-20:
A version number is included in the CI built package names; this must be consistent with the version number reported by `[horace|herbert]_init`

nickbattam-tessella, 2020-01-20:
Historically the version headers were injected with a TortoiseSVN commit hook. 

This has been replaced with the `update_svn_revision_info.m` function in `Horace\admin` called from `make_horace_deployment_kit` on both Herbert and Horace source files.

hsautessella, 2020-03-31:
Following https://github.com/pace-neutrons/Herbert/pull/108, the templates in Herbert have moved. This means we'll need to update the `FindHerbert.cmake` script to find those files in their new directory

PR #175 (2020-04-08): Manage version number for builds and releases
This implements a similar a versioning strategy to https://github.com/pace-neutrons/Herbert/pull/108.

- The version (of format `<major>.<minor>.<patch>`), will be held in the top-level file `VERSION`.
- CMake will read this at configure time and output:
    - `horace_get_build_version.m`
    - `version.h`

   which expose the version number to the Matlab and C++ source.
- If the build is a developer build (i.e. CMake is run without the `Horace_RELEASE_TYPE=RELEASE` flag), the current Git revision SHA will be appended to the version.
- If CMake has [... cut]
closed 2020-04-08 by hsautessella