pace-neutrons/Horace #43: Create Linux CI build

nickbattam-tessella, 2019-11-13:
Create a Jenkins build and test job for the LINUX build.

Part of #30

nickbattam-tessella, 2019-12-16:
Blocked waiting for plugins to be installed on ANVIL: 
•	Generic Webhook Trigger https://plugins.jenkins.io/generic-webhook-trigger - Allows GitHub to trigger builds
•	httpRequest https://plugins.jenkins.io/http_request - Send GitHub statuses
•	xUnit https://plugins.jenkins.io/xunit - Test parsing and visualisation
•	Rebuilder https://plugins.jenkins.io/rebuild - Rebuild failed builds
•	Blue Ocean https://plugins.jenkins.io/blueocean - Better pipeline construction and visualisation 
o	  Including Git Pipeline for Blue Ocean and GitHub Pipeline for Blue Ocean
•	Delivery Pipeline https://plugins.jenkins.io/delivery-pipeline-plugin - Multi-job pipeline and deployment visualisation
•	Performance https://plugins.jenkins.io/performance - Display and track benchmarking results
•	Warnings Next Generation https://plugins.jenkins.io/warnings-ng - Parse and display warnings from builds

hsautessella, 2019-12-20:
Removing blocked label, all plugins above except httpRequest and Blue Ocean have been installed. 

There are plans to install blue ocean down the line

closed 2020-01-29 by abuts