pace-neutrons/Horace #48: Investigate: Handle vs Value classes for Python API

nickbattam-tessella, 2019-11-13:
@mducle raised a question as to whether [handle classes](https://uk.mathworks.com/help/matlab/handle-classes.html) are required to create a usable Python API.

This would be a straightforward update to the code, especially as the SQW object is being re-implemented (#21) BUT would have consequences for all existing "power users" as it would change the behaviour of equality checks and assignments.

Related to #27

nickbattam-tessella, 2020-02-19:
Decision to update "main" external classes (e.g. SQW) to be handle classes

mducle, 2020-02-19:
@nickbattam-tessella - thanks for the info, but I've (just!) since done some more investigation. If you compile a standalone library with Matlab R2017b then all the functions/classes in the folder(s) you compile (and all Matlab built-in functions) are accessible in Python using:

```python
>>> import compiled_matlab_library
>>> m = compiled_matlab_library.initialize()
>>> m.eval('eig(rand(3))')
matlab.double([[-0.18794383321803343],[1.7526703425428873],[0.8398632587280312]])
```

However, newer versions of Matlab essentially makes every (Matlab) function in the library private except for specific m-files which you specify at compile time which are made public to the Python interface. Thus using a newer version of Matlab to compile the library we would need to write a single-m-file (handle) class to wrap all the functions in the library anyway (and so in principle it should not matter whether `sqw` is a handle or value class since we would never be exporting it to Python - only the wrapper will be exported to Python).

More info in my comments to Greg's design document...

nickbattam-tessella, 2020-03-18:
Resolved: HANDLE CLASSES ARE NOT REQUIRED

closed 2020-03-18 by nickbattam-tessella