pace-neutrons/Horace #69: Required for #68

nickbattam-tessella, 2019-12-03:


nickbattam-tessella, 2020-01-23:
:blocked: awaiting Alex's return to project

cmarooney-stfc, 2021-01-09:
@abuts Hi Alex, the comment at the top relates this to ticket #154 which is fine, but also couples 154 to Herbert 155, which seems to be an unrelated bug report. Could you have a look please and fix if required? Thanks

abuts, 2021-01-09:
@cmarooney-stfc: Herbert #155 was substantial part of MPI framework design. It was a bug with missing data messages, related to undefined state a filesystem has, which took us (it was extensive discussion what to do about this before you arrived) substantial efforts to address. The bug was mentioned in Herbert #155 but resolved by redesigning messages transfer protocol. So all messages mentioned here are relevant. I suspect not all relevant messages are mentioned.

hsautessella, 2021-01-27:
Is this done @abuts? My understanding was that the communicators are all working, including blocking send/receive, we are only attempting to optimize the serializer which is covered in https://github.com/pace-neutrons/Herbert/pull/274

oerc0122, 2023-03-07:
@abuts Is this done?

closed 2023-06-20 by oerc0122