pace-neutrons/Horace #72: Technology test: pass MATLAB function handle to compiled MATLAB

nickbattam-tessella, 2019-12-03:
Meuphonic will call Horace library functions with MATLAB function handles, specifically a MATLAB wrapper around the Python Euphonic library.

We need to prove calling method works with a compiled Horace (MATLAB) library.

nickbattam-tessella, 2019-12-03:
This is a cross-cutting issue that may be interest to work being done by @mducle @rebeccafair

mducle, 2022-03-21:
Completed.

The code is in two parts - [a Matlab-only library](https://github.com/pace-neutrons/light_python_wrapper/) and [a Python interface](https://github.com/pace-neutrons/horace-euphonic-interface). 

[Usage examples here](https://pace-neutrons.github.io/Horace/v3.6.2/user_guide/Interfacing_with_other_programs.html#euphonic-example).

closed 2022-03-21 by mducle