pace-neutrons/Horace #84: Investigate options for removing duplication in Horace and Herbert build and test

nickbattam-tessella, 2019-12-12:
The Horace and Herbert build pipelines share a number of common elements:
- CMake helper functions
- CPP test framework (googletest) is downloaded and compiled from source
- shared utility functions

This ticket should explore what options are available for removing the duplication and removing the need to repeated compilation of the test library.

Output:
- recommendation of how to share common code and new ticket with details
- recommendation of whether the repeated build of googletest can or should be avoided

nickbattam-tessella, 2019-12-12:
The idea of having Herbert as a submodule of Horace has been considered previously: https://github.com/pace-neutrons/Horace/issues/8

abuts, 2019-12-12:
in favour of the ideas that the Herert is a self-consistent module and the Horace has Herbert as a dependency. As such, all common code (i.e. unit test framework) should be placed in Herbert

nickbattam-tessella, 2019-12-12:
The other option is to extract common (utility) code into a third repository. 

That would allow other PACE projects using `CMake`, `C++`, `googletest`, `MATLAB` to re-use code without depending on all of Herbert.

nickbattam-tessella, 2019-12-12:
@g5t has use for the CMake helper functions, specifically a find_mp equivalent of the find_hdf function to find the Matlab compatible library.

His current solutions work on Linux and Windows but not MacOS

nickbattam-tessella, 2019-12-16:
Linked to #43 #44 #47 #30

nickbattam-tessella, 2020-01-23:
Time boxed activity to document current ideas on the subject

hsautessella, 2020-03-09:
https://github.com/pace-neutrons/Horace/issues/145 goes a fair way in reducing duplication in terms of CMake scripts. Any common CMake utilities now sit in `Herbert/cmake/shared`, the `FindHerbert` script in Horace then adds that directory to the CMake module path so Horace can use them.

hsautessella, 2020-03-16:
@nickbattam-tessella do you think #145 closes this?

hsautessella, 2020-09-01:
Closing since I think #145 covers the purpose of this issue. Common CMake scripts have all been moved into Herbert

closed 2020-09-01 by hsautessella