pace-neutrons/Horace #86: Add post build commands to copy mex files into DLL directory

hsautessella, 2019-12-16:
Currently, when we build the mex libraries, in order to get Matlab to see them we must choose the DLL directory as the CMake binary directory. This is not desirable as we do not want CMake generated files knocking around within the project source.

We would like to produce the mex libraries in an out of source build and then copy them into the desired directory.

For this we will probably want to use [`add_custom_command(${TARGET} POST_BUILD <copy_command>`](https://cmake.org/cmake/help/latest/command/add_custom_command.html). We should write a `horace_add_mex` function that will do this copy whenever a mex is generated.

nickbattam-tessella, 2019-12-16:
This is related to #73 - the processes of creating release bundles will collect all required files from across the source tree into a single release archive (`.tar.gz`/`.zip`)

hsautessella, 2020-01-06:
This is also needed to start on running Matlab tests in Horace and Herbert so I'll start on this now

nickbattam-tessella, 2020-01-23:
Lined to #91

hsautessella, 2020-02-03:
closed by #110

closed 2020-02-03 by hsautessella