pace-neutrons/Horace #111: Fix and re-enable failing Matlab tests

hsautessella, 2020-01-29:
Several Matlab tests were failing on the build servers (some also locally), to get the servers up and running these were disabled in #101. These need to be fixed/re-written and re-enabled. Below is a `.txt` file containing details about the various tests that were failing.

The commits disbaling these tests were: bea56145a279b33188768a4f400fc4949dfa8b82 85eaec5b1801b20f822ea3304f84d77bc75a0b48

The disable tests are:
```
The disabled tests are:

- test_gen_sqw_cylinder::test_gen_sqw_cyl: single assertion disabled
- test_gen_sqw_powder::test_powder_cuts - test function disabled
- test_multifit::test_fif_single_or_array2 - test function disabled
- test_tobyfit::test_tobyfit_refine_crystal_1 - single assertion disabled
- test_sqw - suite disabled
- test_tobyfit - suite disabled
- test_mslice_utilities - suite disabled
```

[matlab_failing_tests.txt](https://github.com/pace-neutrons/Horace/files/4127719/matlab_failing_tests.txt)

abuts, 2020-01-29:
strange list. 
I have just run the tests on iDaaaS machine and see only test_multifit::test_fit_single_or_array2 failing.
is it possible that these test are running long time? The run on iDaaaS machine takes 48min.

hsautessella, 2020-01-29:
Only the `test_sqw` tests were timing out - and that was due to an error with the MPI workers. I've put the error messages for the failing tests in the `matlab_failing_test.txt` file

hsautessella, 2020-03-09:
`test_gen_sqw_cylinder::test_gen_sqw_cyl` and `test_gen_sqw_powder::test_powder_cuts` only seem to fail for me if using mex. If I run `runtests('test_gen_sqw_for_powder')` (from within the `_test` folder) and `herbert_config.use_mex_C = true` then the tests both fail. If I run the same thing with `herbert_config.use_mex_C = false` the tests pass.

If I run `test_gen_sqw_for_powders` through CTest or `validate_horace` then the tests pass.

hsautessella, 2021-01-27:
I can't actually find `test_multifit::test_fit_single_or_array2` anywhere. Has it been removed? I think this ticket is closable now

closed 2021-02-10 by cmarooney-stfc