pace-neutrons/Horace #134: Copy git hooks at build time

hsautessella, 2020-02-27:
Copying the git hooks at runtime was causing errors when calling `horace_init` with a package and `herbert_config.init_tests` was on - the Herbert package does not contain this function. Now there is a well-defined build step for Herbert/Horace we can copy these hooks at build time.

Additionally, the hooks don't seem to work very well. I'm very often forced to bypass them as they tell me my commits are below 10 characters when they are clearly not. As part of this issue the hooks should be fixed.

**Outputs:**
- CMake commands to copy git hooks at configure stage
- Fix the current githooks so they do not thrown erroneous errors

nickbattam-tessella, 2020-02-27:
Alternatively, update the developer documentation and tell them "there is a git hook <here>; you can install it like <this> and it will <force GitHub issue number to be included in all commit messages".
That should probably be in a section where we are discussion how we want external developers to work with our repo -- do we want them to fork or work on PACE/Horace etc?

cmarooney-stfc, 2021-05-17:
As this actually appears to have been merged long ago, marking it done

closed 2021-05-19 by cmarooney-stfc