pace-neutrons/Horace #138: Create thirdparty library repository

nickbattam-tessella, 2020-02-28:
It is necessary to distribute some third party libraries (e.g. mpich, hdf) as these are required but not necessarily available through MATLAB.

Outcomes: 
- [ ] new Git repo for external libraries (e.g. mpich)
- [ ] scripts to build and commit library to repo
- [ ] update to Horace/Herbert CMake to pull dependencies from repo

nickbattam-tessella, 2020-02-28:
Requires an update to the ADR recording the dependency management plan.

hsautessella, 2020-09-03:
As stated in the [dependency management ADR](https://github.com/pace-neutrons/Horace/blob/eca05aed6a6c64b739a1f44158ab42ad63b14ef9/documentation/adr/0010-package-dependencies-in-repo.md), dependencies are currently stored in the main Herbert/Horace repository.

Closing this.

closed 2020-09-03 by hsautessella