pace-neutrons/Horace #199: Move large test data files into SAN storage area

nickbattam-tessella, 2020-04-15:
Move all the large datafiles used by Horace/Herbert from GitHub and other network locations into the SAN storage area.

Related to Issue #141 

Outcome:
- [ ] source and processed data files on SAN storage area
- [ ] unit tests updated to pull required data from SAN storage
- [ ] existing processed test data removed from GitHub

nickbattam-tessella, 2020-04-23:
Unable to access the SAN storage with the local Jenkins user on the ANVIL build agents.

Waiting for `anonymous` access to be enabled on the share.

hsautessella, 2020-05-22:
SAN now accessible from Jenkins #235

nickbattam-tessella, 2020-05-28:
For Jenkins builds the SAN locations and credentials are pulled from from the Jenkins secrets store; for developer/command line execution the users credentials will be used and SAN location must be sourced from somewhere (e.g. env variable set in a script. local.dev file, ...)