pace-neutrons/Horace #200: MPI parpool jobs fail to complete

nickbattam-tessella, 2020-04-16:
There is an issue with MPI parpool jobs not completing in Windows when they encounter an "expected failure" mode.

The .lock files created by the JobExecutor cannot reliably be removed by the worker when the job configuration has been created resulting in jobs that never complete.

The "expected failures" include:
- job file not written
- job file corrupted
- .lock file locked by the files ystem and cannot be deleted
- no worker starts job file
- worker fails between reading file and starting job execution

The jobs eventually timeout (a long timeout is set to allow the long job execution to complete).

nickbattam-tessella, 2020-04-16:
One possible solution would be to add an extra handshake during the job creation, i.e.

|JobExecutor | Worker |
| --- | --- |
| create lock file | 
| write job description |
| delete lock file |
 |                                 | read job description |
 |                                 | send job-started message|
 | read job-started(1) | |
 |                                 | run job|
 |                                 | write output|
 |                                 | delete job-started message|
| process job completion (2) | |
| delete job-description | |

Separate time outs apply to (1) and (2).
The timeout at (1) should be short (seconds) and the job can be resubmitted quickly if it fails.

abuts, 2020-07-13:
fixed  through #274

closed 2020-07-13 by abuts