pace-neutrons/Horace #227: Extract Horace help from Wiki

nickbattam-tessella, 2020-05-19:
The existing Horace/Herbert documentation is on an internally hosted Wiki.

The discussion around #124 agreed this content will be migrated to rst hosted externally.

Output:
 - [ ]  Existing Wiki markdown extracted and stored in GitHub

nickbattam-tessella, 2020-05-19:
As per #124 the format conversion can be largely automated.

nickbattam-tessella, 2020-05-22:
[wikipages.zip](https://github.com/pace-neutrons/Horace/files/4667081/wikipages.zip)

nickbattam-tessella, 2020-05-22:
Content should follow the style guide (where applicable): https://documentation-style-guide-sphinx.readthedocs.io/en/latest/style-guide.html

oerc0122, 2020-11-09:
This should now be mostly fixed in the `horace-docs` repo  [https://github.com/pace-neutrons/horace-docs](url)

closed 2020-11-18 by oerc0122