pace-neutrons/Horace #233: blocking-messages - create C++ serializer

nickbattam-tessella, 2020-05-21:
Create a C++ serializer to manage blocking message content

part of #69 

Output
- [ ] ability to convert data object to message body
- [ ] ability to parse message body to data object
- [ ] unit tests demonstrating reversibility
- [ ] documentation of data format

nickbattam-tessella, 2020-05-21:
Do we need to develop a custom serializer, or can we use existing libraries to provide this functionality?

abuts, 2020-05-21:
probably need to write our own because of the request of everything working the same way with or without mex code. CPP mpi would not work without so may be not so important in this case....

abuts, 2020-05-21:
I think rewriting hlp serializer in C++ would be beneficial and not that difficult as there is Matlab prototype

abuts, 2020-05-21:
and external library need to be maintained. It was undocumented access to Matlab serializar in previous Matlab versions -- unfortunately not here any more

cmarooney-stfc, 2021-01-24:
@oerc0122 Hi Jacob - noticing that none of the boxes in the description are ticked. Could you ensure they are updated or corrected as appropriate please. Thanks

closed 2021-02-10 by cmarooney-stfc