pace-neutrons/Horace #245: Review Horace documentation

nickbattam-tessella, 2020-06-02:
The Horace documentation will be updated and extended as part of the PACE project.

This ticket is to review the current content, considering:
- is is complete and correct
- are any examples useful for the new Horace
- identifying areas where additional help needs to be written

Relates to issue #124, #203 

Outputs
- [ ] List of required document text updates
- [ ] List of example code fragments that need to be updated for new Horace

nickbattam-tessella, 2020-07-01:
A broader review of documentation is being carried out by @g5t with the excitation group. This is blocked until that activity is completed.

cmarooney-stfc, 2023-10-03:
added @oerc0122  as he is otherwise dealing with documentation - probably this ticket is superseded and can be closed

cmarooney-stfc, 2023-10-03:
superseded by #1238

closed 2023-10-03 by cmarooney-stfc