pace-neutrons/Horace #250: PixelData - Support large-file operation of Horace

nickbattam-tessella, 2020-06-02:
Update all remaining uses of the PixelData object to correctly manage the large-data use-case.

Part of #224 

Output
- [ ] All uses of PixelData rewritten to support large datasets
- [ ] Additional issues created and linked for hard updates

nickbattam-tessella, 2020-08-26:
Create linked tickets for each function or discrete code-block to aid tracking

nickbattam-tessella, 2020-12-16:
Closed as superseded by #434: implementation shifted to algorithm focussed priorities

cmarooney-stfc, 2020-12-29:
@nickbattam-tessella @hsautessella  Hi Nick, Harry. Noting that this is in the done column. Appreciating that this has been put on the back burner to get the algorithms in #434 done but the items above #398, #400, #401 seem to have been postponed rather than finished so wondering if Done is the best location for this. thanks Chris

hsautessella, 2021-01-07:
The intention of this ticket was to enable Horace to work with user-facing file-backed objects. We've since agreed that using file-backed objects under-the-hood and hidden from users was simpler (see the [relevant ADR](https://github.com/pace-neutrons/Horace/blob/master/documentation/adr/0018-algorithm-migration-to-paged-files.md)). 

All children of this ticket should be closed (or "_Done_"), not postponed.

closed 2021-01-13 by nickbattam-tessella