pace-neutrons/Horace #258: Document build/release process

nickbattam-tessella, 2020-06-08:
Extend the build/release pipeline document (https://github.com/pace-neutrons/Horace/blob/master/documentation/smg/10_build_release_process.md) to include additional details

Outcome
- [ ] documentation on the command line build process
- [ ] documentation about how to release a jenkins build
- [ ] documentation on where to upload build artifacts on GitHub as <releases>
- [ ] add note to README.md

hsautessella, 2021-01-27:
We have [12_deploying_releases_to_GitHub.md](https://github.com/pace-neutrons/Horace/blob/master/documentation/smg/12_deploying_releases_to_GitHub.md) describing the release process now

hsautessella, 2021-01-27:
> documentation on the command line build process

Is covered by [SMG 04](https://github.com/pace-neutrons/Horace/blob/master/documentation/smg/04_building_with_cmake.md).

>  documentation about how to release a jenkins build
documentation on where to upload build artifacts on GitHub as 

Are covered by [SMG 12](https://github.com/pace-neutrons/Horace/blob/master/documentation/smg/12_deploying_releases_to_GitHub.md).

> add note to README.md

Does this mean the top-level README? What is it exactly we want to put in the README? In any case, I think any changes should be under a separate ticket, this one is too non-specific and will likely never be picked up.

closed 2021-02-10 by cmarooney-stfc