pace-neutrons/Horace #268: Move source data files from developer machine to SAN

nickbattam-tessella, 2020-06-24:
There are a number of "source data" files located on Toby Perring's machine and used in "-setup" operations for Horace tests.
These should be moved to the SAN storage area and tests updated accordingly.

Related to #199

nickbattam-tessella, 2020-06-24:
- `Fe\sqw\Fe_ei787.sqw`
- `Fe\9cards_4_4to1.par`

cmarooney-stfc, 2020-09-30:
@tgperring Hi Toby - I'm picking up this as a background task in the backlog of the main path. Nick says the aim is to (a) identify source data files on your machine (the 24/6 comment above is apparently a non-exhaustive list) (b) move them to somewhere public (c) if that public location is not the SAN disk, then move them to the SAN disk. Could you assist with steps (a) and (b) please?
Thanks Chris

oerc0122, 2023-06-20:
Files have been moved