pace-neutrons/Horace #312: Remove support for legacy multifit and tobyfit

nickbattam-tessella, 2020-07-29:
The code currently includes "legacy" implementations of Tobyfit and Multifit.
Both are clearly marked with a deprecated warning:
```
%   If you are still using the legacy version then it is strongly recommended
%   that you change to the new operation. Help for the legacy operation can
%   be <a href="matlab:help('sqw/multifit_legacy_sqw');">found here</a>]
```

These, and the supporting infrastructure should be removed from the codebase.

nickbattam-tessella, 2020-07-29:
Removal confirmed by Toby via email on 29-Jul-2020

nickbattam-tessella, 2020-09-23:
These will be removed as (new) fit are moved into the new-style class SQW folder.

Part of #288

cmarooney-stfc, 2020-12-29:
@nickbattam-tessella Hi Nick, Checking this should still be tagged blocked since you have moved it into In Progress. Thanks Chris

cmarooney-stfc, 2021-01-09:
@nickbattam-tessella Hi Nick, noting you have moved this back into sprint backlog, but also that PR 313 is also outstanding albeit marked do-not-merge. What is the status of the PR? thanks

cmarooney-stfc, 2021-11-01:
MOved this ticket back to sprint backlog as the complete removal of legacy has not yet been done. What has been done is in pace-neutrons/Herbert#392 which is a PR to deprecate use of legacy at the point of use. The larger removal task will be prioritised later

PR #1068 (2023-07-03): Remove multifit legacy
Fixed #312