pace-neutrons/Horace #345: Update permute_pix_array for file-backed pixel data

hsautessella, 2020-09-08:
The `permute_pix_array` function needs to be updated such that it does not assume that all pixel data is held in memory.

An example of how to loop through pixel data is given in: https://github.com/pace-neutrons/Horace/blob/master/horace_core/sqw/PixelData/%40PixelData/private/compute_bin_data_mex_.m.

Part of #250 

**Outputs:**
- `permute_pix_array` updated to use `PixelData` objects
- `permute_pix_array` algorithm works if not all pixel data can be held in memory at one time
- tests verifying `permute_pix_array` works with file-backed pixel data

nickbattam-tessella, 2020-09-09:
This functionality is in slice, and the function is not required any more.

cmarooney-stfc, 2020-09-14:
Closed following PR merge to fix

closed 2020-09-14 by cmarooney-stfc