pace-neutrons/Horace #363: Update PR-* pipelines to merge base PR branch

hsautessella, 2020-09-15:
At the moment the PR-* pipelines always merge the pull request branch with master. This means, when we're building PRs to be merged into branches other than master, we're building the wrong thing.

This is fairly straightforward to do, but requires updating the pipelines in the GUI. The key `$.pull_request.base.ref` needs to be added to the Generic Webhook Trigger to extract the base branch name from the pull-request payload. 
[image]

The checkout settings then need to be updated to use the extracted base branch.
[image]

I've added a [test pipeline](https://anvil.softeng-support.ac.uk:444/jenkins/job/PACE-neutrons/job/Horace/job/PR-Scientific-Linux-7-TESTER-2018b/configure) that includes the updated settings.

hsautessella, 2020-09-15:
A problem with this is the Herbert branch that is used. At the moment, Herbert master is pulled, however there is no guarantee that Herbert master will be compatible with the base branch of the PR.

hsautessella, 2020-09-15:
Options:

- We can use labels in the GitHub pull request to dictate which Herbert branch to pull.
  - for example a PR with label `Herbert_3_4` will pull Herbert branch `rel_3_4`
  - a pull request with no label will pull Herbert master
This can be achieved using the generic webhook plugin, and parsing the labels sent in the pull request's JSON. 

The `$.pull_request.labels[*].name` pattern will extract a list of labels from the webhook JSON. An example output being:
```
[
   "Herbert_3_4",
   "Label we don't want"
]
```

There is a "filter" option on the plugin that allows one to strip characters from string parsed from the JSON. We will need to be able to anti-regex the `Herbert_*` label, to remove the characters in the string we don't want. In the example above we want to match everything _except_ `Herbert_3_4`, so that only `Herbert_3_4` remains after the "filter" does its work.

nickbattam-tessella, 2020-09-16:
Quick test regex finds: `^((?!Herbert_[a-zA-Z0-9_]).)*` matches all except `Herbert_3_4",` in this test block

[
   "Unwanted",
   "**Herbert_3_4",**
   "Label we don't want"
]

(using https://www.regextester.com)

nickbattam-tessella, 2020-09-17:
The discussion above would have changed the wrong branch -- that is the base (Horace) branch the PR is merged into not the Herbert branch pulled.

The use of PR labels is the simplest solution, however the regex implementation is not required.
The branch label can be extracted at script level from the list of all labels included in the PR metadata. Pulling the Herbert source is a script-based action already.

This allows an additional checks of the uniqueness of the Herbert branch-tag label and is much simpler.