pace-neutrons/Horace #373: Pipeline updates have added Branch description to Jenkins nightly builds

nickbattam-tessella, 2020-09-21:
The pipeline updates #363 have changed the build descriptions in Jenkins.

Currently: nightly build pipelines are all described as `Branch: master`
Previously: nightly builds did not show a `Branch:` description, these only appeared on the `branch_` pipelines

The branch description was added to the branch build pipelines as the same pipeline is reused by any branch being built. This enables developers to track a particular branch build history, or to identify which of the two running branch builds is the one they triggered.

The nightly/master pipeline is always building the same branch, so this information is just noise in the UI.

nickbattam-tessella, 2020-09-21:
[image]

oerc0122, 2023-06-20:
Closed as won't fix