pace-neutrons/Horace #383: Herbert bug #234, for noisify error interpretation and interface extension

cmarooney-stfc, 2020-09-23:
Herbert bug #( 234 )  reports bug for array size comparison and use of error as standard deviation rather than variance. It also extends the argument list to add arg 4 = max signal value.

hsautessella, 2020-10-06:
Same issue as https://github.com/pace-neutrons/Herbert/issues/234 which was closed by https://github.com/pace-neutrons/Herbert/pull/236#discussion_r495842703, so closing this. 

If I'm mistaken, please feel free to re-open, Chris.

closed 2020-10-06 by hsautessella

cmarooney-stfc, 2020-10-07:
Thanks Harry this looks fine. Just commenting that for the links in your comment above, the first one is an issue, the second a PR. Both referenced with #number. Do we have a good way of distinguishing issues and PRs when doing this?

hsautessella, 2020-10-07:
Not that I know of, GitHub doesn't seem to differentiate pull requests from issues. If you have an issue URL e.g. `https://github.com/pace-neutrons/Horace/issues/383` you can equally access that via `https://github.com/pace-neutrons/Horace/pull/383` and vice versa for pull request URLs. I suspect pull requests are just issue tickets with a Git patch linked to them.

The best I can think of is to start prefixing links to pull requests with "PR-"? e.g. PR-https://github.com/pace-neutrons/Herbert/pull/236