pace-neutrons/Horace #390: Test failing in Horace release branch when using Herbert release branch

hsautessella, 2020-10-01:
Running tests on Horace branch `rel_3_4` (4a5c3184d7246a6c33589e447d7868d670f8ec4b) and using Herbert branch `rel_3_4` (https://github.com/pace-neutrons/Herbert/commit/7c6fcf85caeeef40b6dd772d9ddf6d107db3300e) results in some test failures.

These are the branches the `3.4.x` releases are based on, so these tests should be fixed or it could stop us generating more releases. 

A job running the tests with the two release branches can be found [here](https://anvil.softeng-support.ac.uk:444/jenkins/job/PACE-neutrons/job/Horace/job/Release-Scientific-Linux-7-2018b/6/). Tests failing are:

- projectroot._test.Matlab.test_tobyfit
- projectroot._test.Matlab.test_tobyfit/test_let
- projectroot._test.Matlab.test_gen_sqw_workflow/test_gen_sqw_accumulate_sqw_herbert
- projectroot._test.Matlab.test_gen_sqw_workflow/test_gen_sqw_accumulate_sqw_parpool

**Outputs:**
- Relevant changes made to Herbert/Horace release `rel_3_4` branches such that tests pass when the two branches are run together.

_Note: When opening a pull request to merge changes into Horace branch `rel_3_4`, be sure to put label `Herbert_rel_3_4` on the pull request ticket. This will ensure the pull request is built using Herbert branch `rel_3_4`._

abuts, 2020-10-01:
this was known issue fixed on Herbert Master and Horace Master (not sure about Tobyfit)

hsautessella, 2020-10-01:
Should the fixes for the issue(s) be merged into the release branches?

abuts, 2020-10-01:
the fixes are in master. The merging was intended in pre-release branches. To do easy fix one should be starting from rel_3_4 and I've started from master. Herbert have not diverged much so we can merge prerelease to master and fix all MPI issues and move release tag for Herbert. For Horace the only issue is the failing of the tests due to time-outs. You was playing with time-out changes but I have not looked into what was done there.

oerc0122, 2020-12-03:
Still an issue with 3.4, fixed in 3.5 - Closing following discussion.

closed 2020-12-03 by oerc0122