pace-neutrons/Horace #405: Add notification when dataset is too large for in-memory operation

nickbattam-tessella, 2020-10-13:
A the point we switch from in-memory to paged operation the user will experiences a sudden drop in performance.

A notification should be output to the console at this point to inform them what has happened and why.

The specific text will need to be agreed but I propose it should include information like
i)	that the data set is larger than the currently configured page, 
ii)	the page size can be changed <how>, 
iii)	point the user to some online documentation which will have written documenting advanced optimization techniques, i.e. the page-size/performance curve indicating how to select a good point on it for the target hardware.

oerc0122, 2023-03-07:
Not sure we want this anymore, @abus @cmarooney-stfc discuss?

oerc0122, 2023-06-20:
Done with filebased refactor

closed 2023-06-20 by oerc0122