pace-neutrons/Horace #434: Implement file-backed versions of algorithms

hsautessella, 2020-11-05:
A set of Horace algorithms needs to be updated to work with file-backed pixel data. The API for the algorithms should follow the design decisions laid out in the ADR written in https://github.com/pace-neutrons/Horace/pull/416.

File-backed `PixelData` objects should be used when paging over pixel pages. For performance reasons the implementations should try and minimize the number of loops over the data, as these operations are expensive. Care should also be taken to not "leak" file-backed objects and expose them to users.

**Outputs:**

File-backed implementations of the following algorithms:

- [ ] `gen_sqw`
- [ ] `cut`
- [ ] `symmetrise`
- [ ] `multifit`
- [ ] `tobyfit`
- [ ] `func_eval`
- [ ] `sqw_eval`

hsautessella, 2020-11-05:
Blocked by https://github.com/pace-neutrons/Horace/issues/435.

nickbattam-tessella, 2020-12-16:
supersedes #250

closed 2023-07-05 by oerc0122