pace-neutrons/Horace #455: Documentation Jenkins Pipeline

oerc0122, 2020-11-20:
- [ ] Nightly (unstable) & Release (stable) -> Push
- [ ] Release docs separately in release pipeline as separate zip artifact
- [ ] Build docs at release draft for review
- [ ] Create zipped HTML artifact for checking MRs
- [ ] On each PR build and run docs w/ tests
- [ ] Update documentation version no on release
- [ ] Have stable (last github release), unstable (master [Nightly]), version (e.g. 3.5.1)
- [ ] Stable link through HTML Redirect

cmarooney-stfc, 2021-01-25:
@oerc0122 Hi Jacob - this is in the sprint backlog but I can see some work items have been done. Should this be in "In Progress"?

oerc0122, 2021-02-18:
Already have doc tests issue #458.

cmarooney-stfc, 2021-02-21:
@oerc0122 Hi Jacob, noting that the "On each PR build and run docs \w tests" item is not ticked. SHould this be still in progress? thanks Chris

oerc0122, 2021-02-22:
No, as noted in my closing comment, that is documented under issue #458 , and should be considered a separate issue in the Epic.

cmarooney-stfc, 2021-02-22:
True, but just looking at the issue description, it's not obvious to go down and find your comment. I've added "item now moved to separate ticket" in the issue description - please could you check this is ok? thanks

closed 2021-02-24 by cmarooney-stfc