pace-neutrons/Horace #469: Create API for cluster management

nickbattam-tessella, 2020-11-30:
Create APIs to manage clusters for use by the MPI framework.

If the user is responsible for managing the cluster:
- the startup must be executed before the first job submission
- the shutdown executed after the final job submission 

If the cluster is centrally managed, a cluster wrapper will be created allowing this to be passed to the Job submission.

The code required to start-up and shutdown the three implemented clusters is already included in the Job submission steps, which will be modified to accept a running cluster wrapper as an argument.

Related to #468

nickbattam-tessella, 2020-11-30:
Three distinct use cases:
- in-script call
```
startup
submit(job 1)
submit(job 2)
...
submit(job n)
shutdown
```
- global cluster startup leaving the cluster available on a background thread
```
startup   |
[ ... ]   | submit(job)
          | submit(job)
```

- externally managed cluster
```
attach-to-running-cluster()
submit(job 1)
submit(job 2)
...
submit(job n)
```

abuts, 2020-11-30:
I believe, all this stuff is already there, except `startup` and `submit` and `attach` are currently executed within `summit_job` method.  This ticket is rather introduction to #468 -- not sure any changes to code are necessary

abuts, 2021-01-26:
The situation have changed from the time the ticket is formulated. The iDaaaS team promises to provide us with cluster management system, based on Slurm . As mentioned in the document, developed under #506, this is much more preferred solution then creating cluster based on custom API. If any other options should should be preferred, it should be reflected under #506. The current API for controlling cluster is described in [MPI Framework document](https://github.com/pace-neutrons/Horace/blob/master/documentation/add/04_mpi_framework.md). As this API already works with 3 frameworks and 3 types of clusters, the description there is sufficient. 

The actual `ClusterWrapper`  for Slurm sluster development should be postponed until a Slurm cluster is there.

cmarooney-stfc, 2021-02-21:
@abuts Hi Alex, thanks for this note. Toby will be asking about progress on providing the cluster and slurm. Are we still waiting for the providers to give us something or are we ready to go forward with the #506 proposals? If we're still waiting, who are we waiting on? thanks Chris

cmarooney-stfc, 2021-02-21:
@abuts Also, should we close this ticket now if we're following #506?

oerc0122, 2023-06-20:
Done with slurm commands revision

closed 2023-06-20 by oerc0122