pace-neutrons/Horace #473: Create Release Process flowcharts

nickbattam-tessella, 2020-12-02:
Add diagrams to the SMG illustrating the development processes.

Update Process SMGs (`10_build_release_process.md', `12_deploying_releases_to_GitHub.md`) .

This will make the process clearer and easier to follow and check.

- [ ] Dev cycle
- [ ] Jenkins pipelines, triggers and dependencies
- [ ] Release/deploy process

cmarooney-stfc, 2020-12-29:
@hsautessella Hi Harry, Checking why this is in Review with only one of 3 boxes ticked. Is the intention to go back to in progress? Also noting (although it is more general than this ticket) that it refers to SMG_10 and there are *two* SMG_10s - I will also bring this up separately. Thanks Chris

cmarooney-stfc, 2021-01-24:
@hsautessella @oerc0122 Hi both - checking the status of the unticked Dev cycle box in the description, as this is now in Done. Thanks

closed 2021-05-19 by cmarooney-stfc