pace-neutrons/Horace #474: Add Linting to MATLAB code

nickbattam-tessella, 2020-12-03:
The CppCheck is run during the C build process to perform static analysis on the C source and results reported in the Jenkins dashboard.

MATLAB provides the mlint tool which should be added to the Jenkins build pipeline to identify and report potential code issues and allow these issues to be monitored and tracked.

hsautessella, 2021-02-02:
We should use the [Warnings Next Gen](https://github.com/jenkinsci/warnings-ng-plugin) plugin to display the linter's results on Jenkins. Matlab's `checkcode` function (which is the recommended way of running `mlint`) generates a `struct` containing results; there's no option to output the report to an `xml`/`json` file that "Warnings Next Gen" can parse. Because there's no standard file format for `mlint` results, there's no standard parser that the Jenkins plugin can use.

A large part of this task is to convert the `struct` outputted by `checkcode` into the Warnings Next Gen native `xml` or `JSON` format - see https://github.com/jenkinsci/warnings-ng-plugin/blob/master/doc/Documentation.md#export-your-issues-into-a-supported-format

nickbattam-tessella, 2021-02-10:
Related to #547

closed 2021-02-24 by cmarooney-stfc