pace-neutrons/Horace #486: Create running Python/compiled MATLAB demo

nickbattam-tessella, 2020-12-15:
Confirm that all pieces are in place and work correctly together to enable a compiled Horace to run behind the Python wrappers

- [ ] compiled Horace
- [ ] script executing horace_version
- [ ] script load SQW from file 
- [ ] script executing a cut
- [ ] script executing a fit

mducle, 2020-12-15:
So at a minimum you would need to configure `cmake` to use [`mcc`](https://www.mathworks.com/help/compiler/mcc.html) to compile the Matlab code. An example is [here](https://github.com/mducle/hugo/blob/master/src/CMakeLists.txt#L54). You can use the `-W python:<package_name>` option to create a Python package which you can then do `import package_name; m = package_name.initialize()`. Then every file you explicitly specify in the `mcc` commandline is available as a method of the `m` object / `package_name` (Python) class.

Unfortunately you can only pass it _function_ mfiles directly so that they can be methods of the `package_name` class - you cannot pass it `classdef` m-files or class folders. That's why I use the `-a` flag to include to add the full `herbert_core` and `horace_core` folders to the compiled archive. However the functions and classes in the `-a` flag cannot be _directly_ accessed from the `package_name` Python class (or `m` object). This is why I use the `call` and `call_method` functions which uses `feval` to access the functions/classes in the included folders. 

If you just use the current Matlab code with `sqw` as an old-style class then you can run `cut_sqw` using `feval` but it will then return you a Python `dict`. You can use the [thin wrapper](https://github.com/mducle/hugo/blob/master/src/thinwrapper.m) I create to wrap this in a `classdef` and it will pass through ok as a class.

-----

Personally I would prefer not to have any of these codes in the main Horace repo. This is because for the Python / compiled Matlab package to work you need to include all dependent Matlab code in the same compressed/encrypted `ctf` ("compiled Matlab") archive. This means at a minimum including all the code in `herbert_core` as well as `horace_core`. If we want to have a "PACE" distribution it would also have to have `spinw` as well... since any Matlab code which is not in the archive cannot "talk" to each other (only to Python) - e.g. you *cannot* use `spinw` in `multifit` or `sqw_eval` if you do not include `spinw` in the same compiled archive as Horace/Herbert. 

My preferred solution would be to have a separate package/repo, maybe called `pace` or `pypace` or `pace_python` or `pacy` which is a rebranded [`hugo`](https://github.com/mducle/hugo) which would pull in all the Matlab dependencies and compile them, and then also have the necessary Python wrapper code to make the syntax nicer.

mducle, 2021-01-22:
A repository for this has been created: https://github.com/pace-neutrons/pace-python/

closed 2022-03-21 by mducle