pace-neutrons/Horace #487: Demonstrate Horace / Euphonic integration

nickbattam-tessella, 2020-12-15:
Create a simple script to demonstrate the basic Euphonic workflow:
- call Euphonic with modelling parameters and get a SQW object return value
- compare with SQW object containing data from matching sample

Output
- [ ] Script running basic Horace/Euphonic workflow
- [ ] New issues tickets raise for any issues identified

mducle, 2020-12-15:
So this can be done with the current [`horace-euphonic-interface`](https://horace-euphonic-interface.readthedocs.io/en/latest/#usage) using `disp2sqw_eval`. To use it with `multifit_sqw` or `tobyfit` use:

```matlab
kk = multifit_sqw(w1);
kk = kk.set_fun(@disp2sqw, {@euphonic_sf, {pars, scattering_lengths, opts}, [fwhm]});
wsim = kk.simulate();
```

where `w1` is an `sqw` object.

I've created a pull request for an alternative object-oriented interface: https://github.com/pace-neutrons/horace-euphonic-interface/pull/3

nickbattam-tessella, 2020-12-16:
Requires: example data set and script/workflow from Euphonic team

cmarooney-stfc, 2021-01-05:
@mducle Noticing that in your comment on 15/12/20 this is a multifit simulation of Euphonic. We have discussed differences between SpinW and Euphonic in fitting e.g. SpinW can explore parameter space whereas Euphonic works with a fixed set of CASTEP solution parameters. Hence checking what multifit does when used with Euphonic - is it just delivering a single CASTEP solution in sqw form?

mducle, 2021-01-05:
@cmarooney-stfc Yes, in this case using `multifit_sqw` does add anything compared to using `disp2sqw_eval`. The reason I put this in is because `tobyfit` uses the `multifit` syntax and not the `disp2sqw_eval` syntax - so to do resolution convolution we have to use this syntax. I just copied and pasted by example script which uses `multifit_sqw` because it's faster to evaluate during the test.

cmarooney-stfc, 2021-01-05:
@mducle Thanks for this. Checking that I understand. Even though Euphonic will not do parameter space exploration we would still wish to transform the theoretical CASTEP solution into a  broadened dispersion relation that allows for instrumental broadening of the theoretical values - which is resolution convolution. So the single sqw object we get out of Euphonic with the instrument broadening will have to be generated by Tobyfit and hence use the syntax you employ here.

cmarooney-stfc, 2023-09-20:
@oerc0122 Hi Jacob as you are discussing this with Adam, checking what you think the status of this is

closed 2023-10-03 by cmarooney-stfc