pace-neutrons/Horace #491: Remove old SQW/DnD classes

nickbattam-tessella, 2020-12-17:
The new SQW class has been implemented in parallel with the old SQW class (renamed sqw_old/dnd_old).

Depends on #490 

Outcome:
- [ ] No references to sqw_old in the code
- [ ] No references to d0d_old in the code
- [ ] No references to d1d_old in the code
- [ ] No references to d2d_old in the code
- [ ] No references to d3d_old in the code
- [ ] No references to d4d_old in the code

cmarooney-stfc, 2021-05-17:
Following merge of the new sqw/dnd objects (see ticket #288, now done), checking for continued use of _olds in the code. 

Instances found:
- test_change_crystal_2 - skipped test section which also needs restoring
- test_read_ascii_data - just remove comment to previous read call
- test_rebin (test_rebin_d1d) - variables labelled _old although the may well now be full of new-style objects
- horace_core/herbert/@d*d_old and @sqw_old

abuts, 2022-07-22:
done as part of #810

closed 2022-08-10 by cmarooney-stfc