pace-neutrons/Horace #500: mask_ranom_pixels does not work

abuts, 2021-01-07:
User reported bug with Horace3.4 pixels masking
Should be fixed

`Hi Alex,
 
Here is the script I was using on IDAaaS, just takes a cut and then tries to mask 90% of pixels, with the latter giving the error I showed you before.
Namely:
cc_small=mask_random_fraction_pixels(cc,0.1);

 

Error using PixelData/set.data (line 465)

Cannot set pixel data, invalid dimensions. Axis 2 must have num elements matching current pagesize (22933292), found '2293329'.

Error in sqw/mask_pixels (line 85)

wout.data.pix.data=win.data.pix.get_pixels(mask_array).data;

Error in sqw/mask_random_fraction_pixels (line 69)   

wout(i) = mask_pixels(win(i),mask0);

 

Russell
`
`
sqw_file='/instrument/MAPS/RBNumber/RB2010594/ei50_horace_base_psi_sum.sqw'; 
proj.u=[1,1,0]; proj.v=[0,0,1]; proj.type='rrr';proj.nonorthogonal=true;
proj1.u=[1,0,0];proj1.v=[0,0,1];proj1.type='rrr';
proj2.u=[1,0,0];proj2.v=[-0.5,1,0];proj2.w=[0,0,1];proj2.type='rrr';proj2.nonorthogonal=true;

cc=cut_sqw(sqw_file,proj,[-1,0.05,1],[-5,5],[-0.5,0.5],[0,0.4,30]);
plot(compact(cc))

%Size of pix array is 22933292. Mask 90% of pixels randomly to generate
%smaller object for faster fitting.

%Mask 90% of pixels, to speed up.
cc_small=mask_random_fraction_pixels(cc,0.1);

`

hsautessella, 2021-01-08:
I'm a bit confused because v3.4 does not have `PixelData` objects. Are we sure he's not using v3.5?

cmarooney-stfc, 2021-01-09:
@hsautessella Hi Harry. It's unclear why this is in review/QA. I see your query on the versioning but I thought it should stay in "In Progress" until that is resolved. In the old days I would have this in a category "Need More Info" which is different from Review=Fixed pending testing it now works. Have we done an initial test on Russell's data and has Russell been copied into the loop?

cmarooney-stfc, 2021-01-09:
@hsautessella Hi again. I have now seen PR 501, so sorry, missed the connection. Should we be putting the PR number into the ticket as well as the ticket number into the PR? Thanks

hsautessella, 2021-01-11:
When you reference the issue from the PR, a notification is added to the issue (an example from this ticket):
[image]

I agree this can be easy to miss, but I'd be tempted not to add more "process" to opening a PR.

cmarooney-stfc, 2021-01-12:
@hsautessella I think it is extremely easy to miss, because it is in the timeline and unless one always scans the whole event history in the ticket, it may not get clocked. Actually editing the top description puts all the info in an easy to access location. Let's discuss.

cmarooney-stfc, 2021-01-12:
@hsautessella Additionally it's not about process opening the ticket, it's about process updating the ticket to deal with later events.

closed 2021-01-27 by cmarooney-stfc