pace-neutrons/Horace #503: test_gen_sqw_powder::test_powder_cuts failing on rel_3_5

hsautessella, 2021-01-11:
We should investigate how this happened. At the point of release all tests were passing, we should check if there have been any commits into the Horace or Herbert release branch since then that could have broken something. 

```matlab
>> cd ...\Horace\_test\test_gen_sqw_for_powders
>> runtests test_gen_sqw_powder::test_powder_cuts
   ...
===== Test Case Failure =====
Location: 
Name:     test_gen_sqw_powder::test_powder_cuts

...\Herbert\_test\shared\matlab_xunit_ISISextras\@TestCaseWithSave\TestCaseWithSave.m at line 660
...\Herbert\_test\shared\matlab_xunit_ISISextras\@TestCaseWithSave\TestCaseWithSave.m at line 527
...\Horace\_test\test_gen_sqw_for_powders\test_gen_sqw_powder.m at line 126

w1rings.s and w1rings_stored.s: Relative tolerance failure; max. error = 0.2875 at element (105)
```

hsautessella, 2021-01-11:
I've run the test with the 3.5.0 release package, and it passes. The `rel_3_5` branch must have diverged from the release.

hsautessella, 2021-01-11:
Running `validate_horace test_gen_sqw_for_powders` on IDAaaS, the test passes. What Herbert/Horace revisions are running on IDAaaS @abuts?

hsautessella, 2021-01-12:
As above, on IDAaaS, the test passes when running it through `validate_horace test_gen_sqw_for_powders` - however, if you run the test via `runtests test_gen_sqw_powder::test_powder_cuts`  or `runtests test_gen_sqw_powder` or `runtests .` (within the `_test/test_gen_sqw_for_powders` directory) the test fails.

abuts, 2023-02-06:
This has been fixed

closed 2023-02-06 by abuts