pace-neutrons/Horace #507: Migrate DnD API to new classes

nickbattam-tessella, 2021-01-13:
Migrate the full (old) DnD APIs to the new DnD classes, excluding the 'fit' functions.

Follows #490 

Outcome:

- [ ] All SQW methods implemented on new class
- [ ]  Simple tests implemented to verify all methods can be executed

nickbattam-tessella, 2021-01-13:
This ticket includes the removal of identical duplication of SQW and DnD implementations of functions, moving the code to the `SQWDnDBase` class where possible. This should allow the removal of the current implementations in DnD which cast from DnD to SQW, execute the SQW function and cast the result back to the appropriate DnD class

cmarooney-stfc, 2021-02-07:
First checkbox should be DnD

cmarooney-stfc, 2021-02-07:
Also not sure your checkboxes correspond to the first comment after the description

closed 2021-05-07 by cmarooney-stfc