pace-neutrons/Horace #510: Unclear axis ranges in combine sqw.

abuts, 2021-01-13:
Ticket #502 have indicated an unclear peace of code:
in horace_core/utilities/@sqw/combine_sqw.m rows 110-113
```
 for i=1:ndims1
    step=wout.data.p{i}(2)-wout.data.p{i}(1);
    wout.data.p{i}=make_const_bin_boundaries([min_full{i},step,max_full{i}]);
    % TODO: the previous value used here is different from the value,
    % defined by make_const_bin_boundaries. All seems work. 
    % Was those a bug or a feature?
    %[[min_full{i}-step+eps:step:max_full{i}+step-eps]';
```
the axis boundaries were defined as min/max urange (in the meaning of img_range) -/+ step. 
now they are defined as min/max img_range -/+ step/2 
what is correct? Should it just be img_range?

cmarooney-stfc, 2021-01-25:
@abuts Hi Alex - you say that ticket #502 shows this issue - but 502 is a PR. Could you say what the ticket is please? Thanks Chris

abuts, 2021-04-16:
This ticket has been fully fixed by doing #658 as the code does not contains this ranges any more

closed 2021-04-22 by cmarooney-stfc