pace-neutrons/Horace #523: Remove duplicated test data files

nickbattam-tessella, 2021-01-18:
The same test data is included in Herbert and Horace test source and there are some file repeated in multiple test folders within a single project.

All test data should appear only once to keep the (large) repository size under control:
- Data used in multiple test folders should be stored in `[Horace|Herbert]/_test/common_data`
- Data used in both Horace and Herbert tests should be stored in `Herbert/_test/shared_data`

The `shared_data` folder is included in the Herbert build artefact.

Related to #521

nickbattam-tessella, 2021-01-18:
Linked to #199 -- is there a need to store a lot of this test data in Git at all?

oerc0122, 2023-06-20:
No more herbert