pace-neutrons/Horace #526: Add filebacked flag to cut API

hsautessella, 2021-01-19:
To perform out-of-memory cuts, a user must call `cut` with no output arguments and specify an output file:

```matlab
outfile = '/home/user/my_cut.sqw';
cut(sqw_obj, proj, u_lims, v_lims, w_lims, en_lims, outfile)
```

This syntax could be quite confusing, as the function acts in a very different way to when an output argument is specified.

The proposed new syntax for out-of-memory cuts would be:

```matlab
outfile = '/home/user/my_cut.sqw';
sqw_cut_file = cut(sqw_obj, proj, u_lims, v_lims, w_lims, en_lims, '-filebacked', outfile)
strcmp(sqw_cut_file, outfile)  % -> true
```
or
```matlab
sqw_cut_file = cut(sqw_obj, proj, u_lims, v_lims, w_lims, en_lims, '-filebacked', true)
% sqw_cut_file is a temp file with some generated name - we may need to carefully consider how these files are managed
```

Since the old syntax has been released it should be maintained for now, however we should consider adding a warning that the syntax has been deprecated. 

See https://github.com/pace-neutrons/Horace/pull/524 for original proposal/discussion.

mducle, 2021-01-19:
@hsautessella @cmarooney-stfc @oerc0122 @nickbattam-tessella 

Just a comment about the flag syntax, but this might not be the right place for it.

I would like to establish a convention where a minus sign indicates a (`true`) flag, and (`string`, `value`) pairs where the string doesn't have a minus sign indicate (keyword) options. That is, `fun(..., '-someoption')` is the same as `fun(..., 'someoption', true)` which implies that `fun(..., 'someoption', false)` is the default. 

This would mean that we would use `cut(..., '-filebacked')` with the minus or `cut(..., 'filebacked', true)` without the minus... 

What do people think?

mducle, 2021-01-19:
> we may need to carefully consider how these files are managed

Yes. I propose two options (but there might be more!):

1. Keep the syntax as it is and document that it is the user's responsibility to delete the file when they don't need it anymore (or maybe rely on the OS to do it, but that might not be reliable) - maybe add a warning when the command is run like this?
2. Return a object instead of a plain string. The object just wraps the string but its class has a `delete` method which actually deletes the file(s) when the object is deleted.

hsautessella, 2021-01-19:
> I would like to establish a convention where a minus sign indicates a (true) flag, and (string, value) pairs where the string doesn't have a minus sign indicate (keyword) options.

Good spot, I agree. I think this is generally what is done in Horace, and I've just mistyped in the issue description. I personally favour using the `keyword, value` approach rather than adding more `-flags`. My reasoning being that Matlab's `inputParser` supports this and my guess is it would also make translating Python keyword arguments more straightforward.

cmarooney-stfc, 2021-02-21:
@hsautessella Hi Harry, as PR 524 is now merged, can this be closed? thanks

cmarooney-stfc, 2021-02-21:
@hsautessella and indeed put in Review/QA or similar?

cmarooney-stfc, 2021-02-21:
Cross-referencing Duc's comments to #556, #575, #576, #577, #578, 578 where they have been taken forward further

oerc0122, 2023-03-07:
@abuts @cmarooney-stfc  Is this done? To review?

oerc0122, 2023-06-20:
Filebacked refactor renders irrelevant

closed 2023-06-20 by oerc0122