pace-neutrons/Horace #556: Document current and proposed MATLAB / Python API

nickbattam-tessella, 2021-02-10:
A number of different API syntaxes are used for optional and flag arguments in MATLAB.

These should be reviewed and documented as part of a discussion of how these map to the Python API

cmarooney-stfc, 2021-02-21:
@nickbattam-tessella I have now realised that ticket #575 actually requests the ADR and links to PR 583. I'm presuming that this is the same issue - please correct if not, but for completeness it would be good if this could - if not deleted - be linked through to the other items. Thanks

oerc0122, 2021-02-22:
This issue was regarding developing the docs required the meeting and discussion, not the ADR that came out of it. Have "done"d and added the appropriate linkback from the (now closed) PR

closed 2021-02-24 by cmarooney-stfc