pace-neutrons/Horace #560: Update/remove unreliable equal_to_tol test

hsautessella, 2021-02-11:
`test_using_fraction_argument_is_faster_than_comparing_all_pix` asserts on results of benchmark comparisons, which is causing the test to fail occasionally.

The intention of the test is to check that only a fraction of pixels are being compared in the `equal_to_tol` call. Without mocking there was no obvious way of doing this, so I settled on taking the median of some benchmarks. 

**Outputs:**
- `test_using_fraction_argument_is_faster_than_comparing_all_pix` re-written to make it reliable

nickbattam-tessella, 2021-02-18:
Current implementation compares the average time of 1 rep of 5 iterations, i.e. the average time of 5 executions of the comparison.

This gives times (on my dev machine) of 0.54 sec (full) and 0.45 sec (fraction=0.2)

nickbattam-tessella, 2021-02-18:
Fraction code is in `sqw.equal_to_tol` (moved to `equal_to_tol_internal` in SQW refactor)