pace-neutrons/Horace #567: MLint run on Herbert and Horace in Horace Jenkins builds

hsautessella, 2021-02-12:
In Horace builds we clone (or, in the case of nightlies, copy) Herbert into the Jenkins workspace. We then run MLint in the root of the workspace, which recurses into the Herbert directory. This is artificially inflating the number of MLint warnings that are being reported in Horace builds.

**Output:**
- Horace builds should not report MLint warnings from files in Herbert

oerc0122, 2021-02-15:
This leaves two options, I guess.
1. Report all warnings in Horace so they're in once place
2. Keep separate reports for ease of finding them. 

I would probably say option 2, as I think you're suggesting. 

Solution to 2 would probably be scan
```
string(TOLOWER ${PROJECT_NAME}_core "${PROJECT_NAME}_CORE")
lint_json('${${PROJECT_NAME}_ROOT}/${${PROJECT_NAME}_CORE}')
```

hsautessella, 2021-02-15:
The problem with that is we want to lint code that's not in `horace_core` as well. My instinct was to add an `exclude` flag to the `lint_json` script, i.e. `lint_json('', outfile, 'exclude', 'Herbert/*')`. We can create a CMake variable such as `PACE_MLINT_IGNORE` which we can set in Horace so any Herbert directories are ignored. 

The `exclude` flag would also be useful to stop us getting warnings from external code (such as the xunit test framework).

oerc0122, 2021-02-15:
Good point. Probably a good way to set it up.

closed 2021-02-24 by cmarooney-stfc