pace-neutrons/Horace #574: Hardcode and override Jenkinsfile parameters

hsautessella, 2021-02-16:
At present, the Jenkinsfile sets parameters _and their defaults_ in a `properties([ parameters([ ... ]) ])` block. This is nice because parameters are defined in code and are therefore easy to add/remove from all pipelines. However, if we want to update a default, the change does not apply to the first build that runs the updated Jenkinsfile. The first build updates the parameter definitions on the pipeline, _but only after that build has started_, hence the build does not run with the updated parameter value. 
This can cause confusion, especially if there are other pull requests that are running builds (using the old Jenkinsfile) at similar times: the pipeline parameter definitions will be getting changed back and forth.

A proposed solution to this problem is to make all parameter defaults empty and hard-code default values in the pipeline code. A non-empty parameter value can then override the hard-coded value. This gives us the flexibility of specifying parameters in our builds, whilst avoiding the problem described above.

Something like the following should work:

```groovy
properties([
  parameters([
    string(
      defaultValue: '',
      description: 'The version of CMake to run the build with.',
      name: 'CMAKE_VERSION',
      trim: true
    ),
    ...
  ])
])

if (!"${CMAKE_VERSION}".trim()) {
  def CMAKE_VERSION = "3.19"
}

pipeline {

  ...

  stages {
    stage("Run-CMake") {
      script {
        sh """
          module load cmake/${CMAKE_VERSION}
        """
      }
    }
  }
}
```

oerc0122, 2021-02-16:
I can see this getting very cluttered, very quickly. Is there a way in Groovy to do this programmatically?

hsautessella, 2021-02-16:
A more robust solution, that will not error if a parameter is new, may look like:

```groovy
properties([
  parameters([
    string(
      defaultValue: '',
      description: 'The version of CMake to run the build with.',
      name: 'CMAKE_VERSION',
      trim: true
    ),
    ...
  ])
])

String get_string_param_or_default(String param_name, String default_val) {
    String value = "";
    try {
        value = env."${param_name}";
    } catch (groovy.lang.MissingPropertyException _) { }
    if (!value) {
        value = default_val;
    }
    return value
}

def CMAKE_VERSION = get_string_param_or_default("CMAKE_VERSION", "3.19");

pipeline {

  ...

  stages {
    stage("Run-CMake") {
      script {
        sh """
          module load cmake/${CMAKE_VERSION}
        """
      }
    }
  }
}

```

hsautessella, 2021-02-16:
I would really like to be able to refactor out all our helper functions into a separate file (or a [shared library](https://github.com/pace-neutrons/Horace/issues/381)). I think it should be possible, but haven't had the time to try it yet

closed 2021-02-24 by cmarooney-stfc