pace-neutrons/Horace #597: Paralllel tests intermittently failing

oerc0122, 2021-02-25:
Parallel tests are failing intermittently in both Horace and Herbert due to timeouts or deleted files or other issues.

This inconsistency in testing is resulting in "failed" tests being ignored or obscured on PRs. 

For now, the tests should be skipped until they can be debugged. 

A few examples follow:
```
 Warning:  Can not remove test folder C:\Users\Jenkins\AppData\Roaming\MathWorks\MATLAB\mprogs_config_Windows-10-2018b\folder_twa_FCCGKNPDCT; message: No directories were removed.
  > In try_to_create_folder (line 40)
Lab 2: 
```

```
 Error using receive_message_ (line 10)
Job: "job_nsqw2sqw_test_sqw_3023200118" have been canceled. No message folder exist
Error in MessagesFilebased/receive_message_internal (line 487)
            [ok,err_mess,message] = receive_message_(obj,from_task_id,mess_name,is_blocking);
Error in iMessagesFramework/receive_message (line 367)
            [ok,err_mess,message] = obj.receive_message_internal(from_task_id,mess_name,is_blocking);
Error in wait_started_and_report_>check_receive (line 49)
    [ok,err,mess]=obj.mess_exchange_.receive_message(1,'ready');
Error in wait_started_and_report_ (line 16)
    [obj,mess] =check_receive(obj);
Error in ClusterWrapper/wait_started_and_report (line 246)
            [obj,ok] = wait_started_and_report_(obj,check_time,varargin{:});
Error in send_tasks_to_workers_ (line 61)
[cluster_wrp,ok] = cluster_wrp.wait_started_and_report(obj.task_check_time);
Error in JobDispatcher/start_job (line 182)
            [outputs,n_failed,task_ids,this]=send_tasks_to_workers_(this,...
Error in write_nsqw_to_sqw (line 215)
        [outputs,n_failed,~,job_disp]=job_disp.start_job(...
Error in gen_sqw (line 298)
            [~,pix_range]=write_nsqw_to_sqw (tmp_file, sqw_file,pix_range_present,wnsq_argi{:});
Error in accumulate_sqw (line 136)
[tmp_file,grid_size,pix_range] = gen_sqw (varargin{:}, 'accumulate');
Error in gen_sqw_accumulate_sqw_tests_common/test_accumulate_sqw11456 (line 729)
            [~,grid_size,pix_range]=accumulate_sqw (spe_accum, '', sqw_file_accum,...
Error in test_gen_sqw_accumulate_sqw_parpool/test_accumulate_sqw11456 (line 76)
            test_accumulate_sqw11456@gen_sqw_accumulate_sqw_tests_common(obj,varargin{:});
... Trimmed test suite layers
https://anvil.softeng-support.ac.uk/jenkins/job/PACE-neutrons/job/Horace/job/PR-Windows-10-2018b/746/console

```

hsautessella, 2021-02-25:
It seems that configs are being written to a common place (outside the Jenkins workspace) within pipelines, this may explain some test failures.

Take [this job](https://anvil.softeng-support.ac.uk/jenkins/job/PACE-neutrons/job/Herbert/job/Windows-10-2018b/86/testReport/junit/projectroot/_test/Matlab_test_mpi_test_ParpoolMPI_Framework/) as an example, where `test_ParpoolMPI_Framework` is failing:
-  `finish_task_tester` in Herbert is calling `set_config_folder(config_exchange_folder)`
- `set_config_folder` then calls `try_to_create_folder` which attempts to create a folder within `C:\Users\Jenkins\AppData\Roaming\MathWorks\MATLAB\mprogs_config_Windows-10-2018b`, to check that directory is writeable (so that a config file can be written there).
- That folder contains Herbert configs, which suggests that, if multiple jobs are running on `Windows-10-2018b`, they share a config.

We should try to keep any config folders within the Jenkins workspace. Currently, we attempt to do this by setting the `TMP` environment variable in CTest to somewhere within the build folder (see [here](https://github.com/pace-neutrons/Herbert/pull/99) and [here](https://github.com/pace-neutrons/Horace/pull/140/commits/45e31fcccb898ddcf5fc6da2c7a31dc4eadb03f2)). This is obviously not working properly in some places. We should:
1) try and identify why we are writing configs (and possible parallel exchange files) outside the Jenkins workspace. and or
2) work out a way for tests and/or parallel jobs to not overwrite existing configs

cmarooney-stfc, 2021-04-19:
@abuts Hi Alex. Please could you add a note saying how the associated PRs fix this issue and also reference those PRs in this ticket so we have a 2-way link for the fixes. thanks Chris

closed 2021-05-19 by cmarooney-stfc